OpenTabletDriver/OpenTabletDriver #960: Support Request: XP-Pen G430S & Huion H610PRO V2 undetected

SavvyKat, 2021-04-05:
# Support Request

Both my XP-Pen G430S and Huion H610PRO V2 tablets are not being detected in OTD. 
[Console Pastebin](https://pastebin.com/AX6sA5DZ)
I am aware of issues with XP-Pen G430S #920 #698, but I cannot get my H610PRO V2 tablet to connect either. I previously used hawku drivers; however, I made sure to uninstall the vmulti drivers and such. I'm unsure if that would complicate anything. 
I would prefer to use my G430s, but I'd be happy to connect either of them.

## What I've already tried

Made sure to uninstall any previous drivers.
[Uninstalled corresponding HID devices in device manager. ](https://github.com/OpenTabletDriver/OpenTabletDriver/wiki/Windows-FAQ#my-gaomonhuion-tablet-isnt-detected-but-is-listed-as-supported)
Uninstalled all phantom devices.
Tried looking in the discord for any information, and tried to use the device string reader to identify the tablet.
Lots of troubleshooting with restarting my computer and plugging in my tablets.

## Diagnostic Information

[Diagnostic Dumps.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6259972/Diagnostic.Dumps.zip)
Dumped diagnostics three times; however, they saved as empty files and crashed the GUI. Decided to upload them anyway but I don't see how they would be of any use.

X9VoiD, 2021-04-05:
Are you running OTD inside some form of sandbox or a limited environment? Because `Failed to get info` is a very rare error on Windows. It means that the library we're using (HidSharp) failed to read the characteristics of a device, either because it got denied of access or you're running in an unsupported environment.

This issue is external to OTD and cannot be fixed from within OTD.

Mrcubix, 2021-04-05:
Talking about the issue on discord rn in misc @X9VoiD

SavvyKat, 2021-04-05:
Issue was caused by my Wooting Two keyboard. Unpluggin the keyboard before starting OTD lets my tablets connect just fine. Going to look into reporting this issue to Wooting.

closed 2021-04-05 by SavvyKat