OpenTabletDriver/OpenTabletDriver #964: Not detecting the Gaomon S620

umn1ch, 2021-04-06:
So, when I open OTD it doesn't see my tablet (GAOMON S620). 

To solve this issue I first installed the previous versions of the OTD, but it didn't help

Then I exported the diagnostics and opened it through Visual Studio. In code, I found out that my tablet in the system is called "Gaomon Tablet"  [*screen*](https://imgur.com/a/LNFEPdY). But after reading the diagnostics.json file I presumed that OTD tryna find the device called "GAOMON S620" (there were a string where were written "Trying to find "GAOMON S620"). 

As soon as I saw this I moved to Regedit and renamed my device to actually what OTD was tryna find. [*screen2*](https://imgur.com/a/Pm0T4n2).

Throughout it didn't help so I still got that "No tablet detected"

X9VoiD, 2021-04-06:
Read [FAQ](https://github.com/OpenTabletDriver/OpenTabletDriver/wiki/Windows-FAQ#my-gaomonhuion-tablet-isnt-detected-but-is-listed-as-supported)

umn1ch, 2021-04-06:
that didn't work at all :(

closed 2021-04-06 by umn1ch