OpenTabletDriver/OpenTabletDriver #969: Support Request: Can't start the program, it asks me to install .NET (already installed)

RandomPotato0, 2021-04-08:
# Support Request

## What I've already tried

Mrcubix, 2021-04-08:
you need .NET 5 x64, not x86

x86 = 32 bits
x64 = 64 bits

RandomPotato0, 2021-04-08:
> you need .NET 5 x64, not x86
> 
> x86 = 32 bits
> x64 = 64 bits

I know, i've installed 64 bit version, but i also tried 32 bit just in case, still, neither works

Kraaaait, 2021-04-10:
I had the same issue and fixed it today. Installing dotnet via the install script provided by Microsoft fixed it for me.

Download the script here:
https://dotnet.microsoft.com/download/dotnet/scripts

Open a PowerShell window in the download folder and run this command:
`.\dotnet-install.ps1 --runtime 5.0.202`

RandomPotato0, 2021-04-10:
@Kraaaait I did as you said but it still asks me to install it

InfinityGhost, 2021-04-12:
This isn't an issue with OpenTabletDriver, rather with .NET 5. Closing but leaving conversation unlocked.

closed 2021-04-12 by InfinityGhost