OpenTabletDriver/OpenTabletDriver #972: driver uses too much RAM

asdasdadsds, 2021-04-09:
So I just downloaded the driver and it already uses almost 2 GB of RAM and this number keeps growing. (more than 2 gb now)
help pls 

[image]

adryzz, 2021-04-09:
That's most probably a memory leak.
Does it happen every time or is it random?
Help > Export diagnostics
(If it doesn't help i think i'll need a memory dump to figure out what's going on)

asdasdadsds, 2021-04-09:
[diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6288151/diagnostics.txt)
It happens everytime. When I open the driver it uses about 200 mb but after few minutes uses more than 1 gb and every second this number grows

adryzz, 2021-04-09:
Ok, nothing notable in the log file.
Are you able to make a memory dump and send it here when the size gets around 250MB/500MB/1GB?

[image]

Mrcubix, 2021-04-09:
Memory leak in Eto, not caused by OTD

adryzz, 2021-04-09:
so it's already known to be in Eto?

asdasdadsds, 2021-04-10:
thanks for your help, i already fixed it with older version (5.2.0)

InfinityGhost, 2021-04-12:
> thanks for your help, i already fixed it with older version (5.2.0)

Then this issue is still valid for v0.5.2.3, this should probably stay open.

InfinityGhost, 2021-05-25:
Likely resolved in an Eto package bump. Closing until further notice.

closed 2021-05-25 by InfinityGhost