OpenTabletDriver/OpenTabletDriver #973: can't bind some buttons on windows

BK8000L, 2021-04-09:
## Description

these buttons cannot be set for bind
[image]
when i try
[image]
i used these buttons (mouse 5, mouse 4) in devocub driver without any problems
[image]

Not for this issue: i first learned about this driver half a year ago, and since then until today i could not use it.
First: almost nothing works with admin rights. i have UAC turned off, so the admin rights are always active, and all the buttons in the menu do nothing. this wasn't the case half a year ago, but the new versions behave like this on my pc, and i thought that with newer versions the driver only got worse. solved by https://github.com/OpenTabletDriver/OpenTabletDriver/issues/681#issuecomment-778259028
Second: filters have not worked correctly until today. i installed devocub filters and they were working completely wrong. my cursor kept shaking even with high delay and strength values. today i updated this filter and it changed its name in the list to "TabletDriver ..."(don't remember what after TabletDriver because i can't recreate it), and it worked the same way incorrectly. i thought something broke and removed the driver with the configuration and plugins and installed a clean version, after that the devocub filter began to work correctly, and i was shocked! at first i thought that i accidentally had a devocub driver running. this problem haunted me  for more than six months on different driver versions, OS versions and tablets.
Third: the driver crashed or hung very often. when i tried to save filter settings my tablet cursor just stopped moving. applying settings also often caused a crash.

for problems above i thought that the driver was not finished yet, and did not take the logs because of this 😞
today all but one problem (this issue) resolved on their own. i have no idea why did the problems go away on their own since i haven't updated my OS for several months (20H2).
i hope you can find solutions and prevent such floating problems from occurring.

Mrcubix, 2021-04-09:
For the first issue, you probably understood what happened, OTD which is a user mode programs can't interact with admin programs, except people disable UAC so everything run as admin, enable it, the end.

As for the second issue, the original devocub is broken, and the multiplier is way higher than it seem, this have been fixed in OTD, tho recently, it went into another update and got fixed.

As for you crashes issue, you might need to provide a crash report that can be found using these steps:

- head to event viewer
- windows logs > application
- Look for a .NET Runtime error releated to OTD
- Copy and paste the content of the box in the middle (it contain the exception) here

for your bind issues, for now, you'll have to use the script runner plugin and ahk if you want to use these buttons as more complex bindings are not supported in OTD right now, tho it will be later on.

BK8000L, 2021-04-09:
@Mrcubix
First: bat file works fine, i don't need to enable UAC.
Second: what is original devoсub? i am using the fixed version for windows 20h1, and my cursor behaves exactly the same on OTD and devocub, yesterday i had different behavior with the same settings of both drivers.
Third: the crashes are gone completely, i can't catch them 😞
thanks for the bind hint, i will use ahk 😊

closed 2021-04-09 by BK8000L