OpenTabletDriver/OpenTabletDriver #981: Hawku smoothing filter plugin huge lag

Mrturtlenugget, 2021-04-12:
When ever I use the Hawku smoothing filter plugin I get huge lag spikes even with anti chatter like 14-50ms 

I use Gaomon S620

And the latest version of otd

X9VoiD, 2021-04-12:
What OS? And define "lag spike".

Mrturtlenugget, 2021-04-12:
Windows 10 and lag spike as in like choppy frames and input lag on osu

X9VoiD, 2021-04-12:
Are you using a weak CPU?

Mrturtlenugget, 2021-04-12:
maybe I don't know if I do or not.

Mrturtlenugget, 2021-04-12:
I have a modern cpu

Mrturtlenugget, 2021-04-12:
I use ryzen quad core I don't know which model

X9VoiD, 2021-04-12:
Hawku smoothing and Devocub antichatter is not intended to be used at the same time. The timer API used is managed by Windows, and the filters itself is already optimized to use SIMD. If there's some sort of performance penalty it incurs then it just means your CPU is weak or you have way too many processes or other forms of CPU load in the background. OTD nor TabletDriverFilters have control over this.

closed 2021-04-12 by X9VoiD

AbstractQbit, 2021-04-12:
Not sure why you'd want to add more smoothing to a tablet that already has some in firmware. Smoothing by it's nature makes the cursor feel more laggy and sluggish.