OpenTabletDriver/OpenTabletDriver #990: Add support for XP-Pen 01 V2 with possibly different firmware.

Maximumtix, 2021-04-14:
Different Strings....
| Manufacturer  XP-Pen |
| Product Name   Deco 01 V2 |
| Width      254mm     |
| Height     158.75mm     |
| Pressure Levels  8192|
| LPI     5080     |

[diagnosticandstring.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6310318/diagnosticandstring.zip)
I added notes in string.txt about spaces and tabs. Also when I request string index 200,tablet disconnects. I used otd getstring <string> command to get these strings. 

Currently pen pressure works but I can only use top left quarter of the drawing area.

SapiensAnatis, 2023-05-19:
I would like to take a look at this issue. I think I can still reproduce the problem on the latest release, although I haven't tried building `master` yet.

jamesbt365, 2023-05-19:
This issue still exists because we have no idea how to get around it with the current way otd can detect tablets.

SapiensAnatis, 2023-05-19:
I just cloned `master` with a tablet that experienced this issue on the latest release, and it seems to be fixed? 

I believe the issue is that the tablet is being detected as a regular XP-Pen 01 V2 as opposed to the XP-Pen 01 V2 (Variant 2) and thus it has half the max X / max Y that it should have.

jamesbt365, 2023-05-19:
The issue *still* occurs on some variants, some are technically fixed.

SapiensAnatis, 2023-05-19:
I see. What do you mean when you say it's a limitation of the detection mechanism? 

I've obviously only looked for the code for all of about 10 minutes, but is it because the detection works by looping through all candidate configurations in `Driver.cs`, and so can arrive at an imperfect match (i.e. non-variant) before it would arrive at a config which would be a better match?

X9VoiD, 2023-05-20:
The limitation is not being able to differentiate between tablets of the same VID/PID and string descriptors.

SapiensAnatis, 2023-05-20:
Is there any other device information we could use to differentiate them?

I'm still interested in attempting it even though my particular issue was fixed, but if the 'good first issue' label no longer applies, perhaps it would be better to leave it to someone else.

X9VoiD, 2023-05-20:
It needs changes on the whole "detection scheme" used by OTD. It might be a task not suitable for someone new to the project as it might just touch almost every part of the core project.

For these kinds of devices to work, we either have to abandon the configuration-based detection or allow configurations to be mutable during runtime and somehow tell OTD when and how to make those mutations.

> Is there any other device information we could use to differentiate them?

The differences are present in the firmware's report descriptor. One of the problems is that HidSharp/Core is not reliable at "reconstructing" the report descriptor on Windows platform (Windows does not allow access to the descriptors in "raw" form).