OpenTabletDriver/OpenTabletDriver #992: Plugin Manager increasingly takes longer to load on each refresh

InfinityGhost, 2021-04-14:
## Description

Root cause appears to be as result of the App.Settings property being changed, causing all event hooks to be rehooked, and eventually causes huge slowdowns and/or hangups.

https://github.com/OpenTabletDriver/OpenTabletDriver/blob/a91c751d2ee57a8edf30225d34ab12fd81ba3ad5/OpenTabletDriver.UX/Windows/PluginManagerWindow.cs#L114-L126

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | NixOS Linux
| Software Version | a91c751d2ee57a8edf30225d34ab12fd81ba3ad5

gonX, 2022-06-18:
#801 (which was superseded by #1059 and #1063 which are merged) would've fixed this. Is this still occuring?

gonX, 2022-11-07:
I'm assuming this is no longer occuring given the merged PRs. Please reopen if this is still not the case @InfinityGhost

closed 2022-11-07 by gonX