OpenTabletDriver/OpenTabletDriver #993: Weird Build Error from Arch AUR

PrincessAkira, 2021-04-14:
## Description
There is some Strange Build Error i get on Arch Linux which i attached below

## System Information:
Ryzen 7 2700
5.11.14-147-tkg-upds Kernel
Latest Dotnet Runtime via the Yay AUR
I Freshly installed that System 2 Hours ago from Writing this

## External Stuff:
The Error: https://paste.thighs.live/oqofikadop.rb
Screenshot of my Dotnet Version:
[image]

peet1993, 2021-04-14:
This is not a bug. The reason is that the Symantec Root Certificate got removed recently on Arch, check the pinned comment here: https://aur.archlinux.org/packages/opentabletdriver-git/

InfinityGhost, 2021-04-14:
Duplicate of #948

closed 2021-04-14 by InfinityGhost