OpenTabletDriver/OpenTabletDriver #999: Not functional UI

acedia3, 2021-04-16:
First of all, thanks for the driver! Cured my anxiety after using G640

The problems I am having:
1) "Show window" does nothing, but "Close" works: having to exit the application entirely to reopen the UI.
2) "Open plugin manager" not working at all

Installed using the guide.

| Operating System | Windows 10
| Software Version | Latest (v 0.5.3.1)
| Tablet           | G640

Mrcubix, 2021-04-16:
1. can't reproduce
2. enable UAC to the second points from the top at least

acedia3, 2021-04-16:
Also, after attempting to run the driver as admin it crashes and never launches again unless I reboot

acedia3, 2021-04-16:
> 1. can't reproduce
> 2. enable UAC to the second points from the top at least

Will test, thanks!

acedia3, 2021-04-16:
> 1. can't reproduce
> 2. enable UAC to the second points from the top at least

Great, that solved the issue with plugin manager and also with the "show window" button lmao. Thanks a lot

acedia3, 2021-04-16:
>enable UAC to the second points from the top at least 
solved the issue

closed 2021-04-16 by acedia3