OpenTabletDriver/OpenTabletDriver #1003: Failed to install the package in the AUR

Dcove, 2021-04-17:
## Description
when I tried to install otd with yay, dotnet printed 2 messages over and over again so it is failed to compile. here are the messages:
```
/home/[my user name]/.cache/yay/opentabletdriver-git/src/OpenTabletDriver/OpenTabletDriver.Native/OpenTabletDriver.Native.csproj : error NU3028: Package 'Microsoft.AspNetCore.App.Ref 5.0.0' from source 'https://api.nuget.org/v3/index.json': The author primary signature's timestamp found a chain building issue: UntrustedRoot: self signed certificate in certificate chain [/home/[my user name]/.cache/yay/opentabletdriver-git/src/OpenTabletDriver/OpenTabletDriver.Daemon/OpenTabletDriver.Daemon.csproj]
/home/[my user name]/.cache/yay/opentabletdriver-git/src/OpenTabletDriver/OpenTabletDriver.Native/OpenTabletDriver.Native.csproj : error NU3037: Package 'Microsoft.AspNetCore.App.Ref 5.0.0' from source 'https://api.nuget.org/v3/index.json': The author primary signature validity period has expired. [/home/[my user name]/.cache/yay/opentabletdriver-git/src/OpenTabletDriver/OpenTabletDriver.Daemon/OpenTabletDriver.Daemon.csproj]
```

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | manjaro linux with linux-zen v5.11.14, dotnet sdk v5.0.4 |
| Software Version | the latest in the aur |
| Tablet           | gaomon m106k pro (in fact it is china's edition, the 1060pro)

gonX, 2021-04-17:
Please verify that dotnet-sdk is at least version 5.0.5: `pacman -Qs dotnet-sdk` - if not, please update your system

Dcove, 2021-04-17:
> Please verify that dotnet-sdk is at least version 5.0.5: `pacman -Qs dotnet-sdk` - if not, please update your system

thx, I noticed that the latest version of dotnet-sdk is not included in the default manjaro repository so i upgraded dotnet-sdk by the switching to unstable repository. now it complied succefully.

closed 2021-04-17 by Dcove