OpenTabletDriver/OpenTabletDriver #1008: Display menu bugging out

Ikxi1, 2021-04-18:
## Description
Hello,
I just downloaded this driver and it seems the display menu is kinda bugged.
I think it is because I'm using 1 of my monitors in portrait mode (seen in screenshot).
I don't know how to position the window like this, so that I only have my middle screen in the field.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Windows 10 (latest update just yesterday)
| Software Version | 0.5.3.1
| Tablet           | Wacom CTL-490
[image]

AbstractQbit, 2021-04-18:
Right click on display area editor > Set to display > Choose the one you need

Ikxi1, 2021-04-18:
> 
> 
> Right click on display area editor > Set to display > Choose the one you need

.....yea, I just found out
I didn't really expect that you can actually do that.
Found out accidentally. (maybe include in the start guide)

Thank you!!

closed 2021-04-18 by Ikxi1