OpenTabletDriver/OpenTabletDriver #1011: Support Request:

KurisuDesuru, 2021-04-18:
# Support Request
tablet: gaomon s56k
cursor is not moving
at first he even is not detected then i change gaomon s56k config
tablet detected, but cursor is not moving
os: archlinux

## What I've already tried
fixed ArgumentOutOfRangeException: Value [0-15]

## Diagnostic Information
[diagnostics.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6331816/diagnostics.json.txt)

Mrcubix, 2021-04-18:
provided you followed every steps from the wiki, and checking the console everytime for more errors, it should be working on latest master https://github.com/OpenTabletDriver/OpenTabletDriver/wiki/Linux-FAQ

KurisuDesuru, 2021-04-18:
i try use all faq
[image]
tablet debugger works, but cursor not moving ((((

KurisuDesuru, 2021-04-18:
i get only this warning
[image]

KurisuDesuru, 2021-04-18:
bruh, the problem was solved by installing lib32-libevdev

closed 2021-04-18 by KurisuDesuru

InfinityGhost, 2021-04-18:
Odd that it was looking for a 32 bit libevdev, but if it works its w/e lol