OpenTabletDriver/OpenTabletDriver #1020: dotnet 5.0.202 incompatiblity? (Ubuntu)

PexoHundido69, 2021-04-20:
## Description
When running sudo apt install ./OpenTabletDriver.deb I get:
"The following packages have unmet dependencies:
 opentabletdriver : Depends: dotnet-runtime-5.0 but it is not installable
E: Unable to correct problems, you have held broken packages."

However when running both dotnet --info and dotnet --version it shows 5.0.202 installed.

## System Information:

Runtime Enviroment:

OS Name:     ubuntu
OS Version:  20.04
OS Platform: Linux
RID:         ubuntu.20.04-x64
Base Path:   /snap/dotnet-sdk/120/sdk/5.0.202/

InfinityGhost, 2021-04-20:
Don't use snap, install the runtime through aptitude/dpkg instead.
```sh
# Add the Microsoft Packages repository, use the URL related to your distribution
sudo dpkg -i packages-microsoft-prod.deb
# Install the .NET runtime
sudo apt update; sudo apt install -y apt-transport-https; sudo apt update
sudo apt install -y dotnet-sdk-5.0
```

PexoHundido69, 2021-04-20:
> sudo apt update; sudo apt install -y apt-transport-https; sudo apt update
> sudo apt install -y dotnet-sdk-5.0

E: Unable to locate package dotnet-sdk-5.0
E: Couldn't find any package by glob 'dotnet-sdk-5.0'

InfinityGhost, 2021-04-21:
Did you install the `packages-microsoft-prod.deb`? I realized now that I didn't link it but it can be found here `https://packages.microsoft.com/config/`

PexoHundido69, 2021-04-21:
Yes that's the problem. It is installed.

Ewald-Linux, 2021-04-27:
I'm having the exact same problem. I tried both methods and I'm still getting the same error.

dezgasting, 2021-04-28:
I had exact same issue.
What you need to do is install the packages-microsoft-prod.deb for the version that dotnet 5.0 supports. E.g. I tried Ubuntu, and I had to download (for?) the 20.04 one, because 21.04 doesn't work. Y'all might be downloading the wrong version, try downloading an older one and it will work.

closed 2021-05-25 by InfinityGhost