OpenTabletDriver/OpenTabletDriver #1023: GTK app crashes when it tries to detect the tablet

hadasfada, 2021-04-20:
The daemon works but when I try to open the GTK app it tries to connect to the daemon gets stuck and then crashes.

Exception in Gtk# callback delegate
  Note: Applications can use GLib.ExceptionManager.UnhandledException to handle the exception.
System.NullReferenceException: Object reference not set to an instance of an object.
   at OpenTabletDriver.UX.MainForm.DetectAllTablets()
   at OpenTabletDriver.UX.MainForm.<ConstructMenu>b__11_7(Object sender, EventArgs e)
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__140_0(Object state)
   at Eto.GtkSharp.Forms.GtkSynchronizationContext.<>c__DisplayClass0_0.<Post>b__0()
   at GLib.Idle.IdleProxy.Handler()
   at GLib.ExceptionManager.RaiseUnhandledException(Exception e, Boolean is_terminal)
   at GLib.Idle.IdleProxy.Handler()
   at Eto.GtkSharp.Forms.ApplicationHandler.Run()
   at Eto.Forms.Application.Run(Form mainForm)
   at OpenTabletDriver.UX.App.Run(String platform, String[] args)
   at OpenTabletDriver.UX.Gtk.Program.Main(String[] args)

System Information:
Operating System : openSUSE Tumbleweed
Version : 0.5.3.1
Tablet : Wacom CTL-472

matiidenys, 2021-04-21:
I may be wrong, but seems like it's the same issue we had with OTD on ArchLinux

gonX, 2021-05-15:
Duplicate of #896, should get fixed by #1104

InfinityGhost, 2021-05-25:
Resolved in #1104

closed 2021-05-25 by InfinityGhost