OpenTabletDriver/OpenTabletDriver #1033: Express Buttons not working on Huion H640P

athdesk, 2021-04-22:
## Description
Express Buttons not working on Huion H640P. Also when pushing on one of them, the cursor skips on the upper-left corner of the screen.
I think this is a configuration issue and I might be able to fix it, given some time; if it's ok, I can make a pull request that fixes this.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Linux 5.11.15
| Software Version | OTD 0.5.3.1
| Tablet           | Huion H640P

InfinityGhost, 2021-04-22:
This is most certainly a configuration issue, likely the lack of any advanced parser for the tablet. As of 3b9690dbc2491a7300d112a8ad824a169a24afbe its using the `TabletReportParser` which is designed for only basic tablets.

This tablet likely requires the `UCLogicReportParser` instead, like many other Huion tablets do.

athdesk, 2021-04-22:
I just tried adding an entry in the AuxilaryDeviceIdentifiers field (using the same VID/PID used in the DigitizerIdentifiers entry), handled by UCLogicReportParser, and the buttons now work correctly.
I'll just play around with the debugger to try and fix the cursor skipping when pushing any of them, i'll update this issue as progress is made

X9VoiD, 2021-04-22:
Don't add a new AuxiliaryIdentifier unless it's of a different input/output length

InfinityGhost, 2021-04-22:
Ideally we'd replace the existing digitizer identifier, thats the intended use case for the UCLogicReportParser.

athdesk, 2021-04-22:
Thanks, I'll try and do it this way

athdesk, 2021-04-22:
Using UCLogicReportParser directly in the DigitizerIdentifier works for the express buttons, and limiting the active report range fixes the cursor jumping around.

This is fixed in PR #1034 , I think this issue can be closed

PR #1034 (2021-04-22): Fix Huion H640P Express Keys
Changed configuration for Huion H640P.
This fixes express buttons not working, and cursor jumping

Closes #1033
closed 2021-04-22 by athdesk