OpenTabletDriver/OpenTabletDriver #1041: worked fine before updating, now it doesnt

autoteelar, 2021-04-23:
## Description

not working after update
## System Information:

| Name          | Value |
| ---------------- | ----- |
| Operating System | void linux 
| Software Version | v0.5.3.1
| Tablet           | 470 wacom

was working just fine before updating

A fatal error occurred. The required library libhostfxr.so could not be found.
If this is a self-contained application, that library should exist in [/home/ralts/Documents/OpenTabletDriver/].
If this is a framework-dependent application, install the runtime in the global location [/usr/share/dotnet] or use the DOTNET_ROOT environment variable to specify the runtime location or register the runtime location in [/etc/dotnet/install_location].

The .NET runtime can be found at:
  - https://aka.ms/dotnet-core-applaunch?missing_runtime=true&arch=x64&rid=void-x64&apphost_version=5.0.5
A fatal error occurred. The required library libhostfxr.so could not be found.
If this is a self-contained application, that library should exist in [/home/ralts/Documents/OpenTabletDriver/].
If this is a framework-dependent application, install the runtime in the global location [/usr/share/dotnet] or use the DOTNET_ROOT environment variable to specify the runtime location or register the runtime location in [/etc/dotnet/install_location].

The .NET runtime can be found at:
  - https://aka.ms/dotnet-core-applaunch?missing_runtime=true&arch=x64&rid=void-x64&apphost_version=5.0.5
  
  i have dotnet in my bin folder was already the latest version but i updated it anyways and overwrote the exact same file still didnt work

autoteelar, 2021-04-23:
using the old version works just fine still, but the gui crashes as soon as it loads, the daemon still works but this is an issue for me because i cant change between my drawing and osu presets

gonX, 2021-05-15:
> Exception in Gtk# callback delegate
> Note: Applications can use GLib.ExceptionManager.UnhandledException to handle the exception.
> System.NullReferenceException: Object reference not set to an instance of an object.
> at Gtk.Widget.DrawnSignalCallback(IntPtr inst, IntPtr arg0, IntPtr gch)
> at GLib.ExceptionManager.RaiseUnhandledException(Exception e, Boolean is_terminal)
> at Gtk.Widget.DrawnSignalCallback(IntPtr inst, IntPtr arg0, IntPtr gch)
> at Eto.GtkSharp.Forms.ApplicationHandler.Run()
> at Eto.Forms.Application.Run(Form mainForm)
> at OpenTabletDriver.UX.Gtk.MainClass.Main(String[] args)

This is probably #896 and seems different from your actual issue

InfinityGhost, 2021-07-18:
Stale support request, assuming it has been resolved. This can be reopened in the case this was a mistake.

closed 2021-07-18 by InfinityGhost