OpenTabletDriver/OpenTabletDriver #1045: CTH480 cursor teleporting like crossmap jumps

naenaebabyy, 2021-04-24:
## Description

The cursor jumps to the topleft corner of my screen like im doing crossmap jumps
## System Information:

| Name             | Value |
| ? |? |
| Operating System |
Windows 10
| Software Version |
0.4.1
| Tablet         CTH480|

X9VoiD, 2021-04-24:
Use the latest version of OTD.

naenaebabyy, 2021-04-24:
> Use the latest version of OTD.

I AM WTF

X9VoiD, 2021-04-24:
The latest version is [0.5.3.1](https://github.com/OpenTabletDriver/OpenTabletDriver/releases/tag/v0.5.3.1), not 0.4.1.

naenaebabyy, 2021-04-24:
> The latest version is [0.5.3.1](https://github.com/OpenTabletDriver/OpenTabletDriver/releases/tag/v0.5.3.1), not 0.4.1.

[image]

naenaebabyy, 2021-04-24:
> > The latest version is [0.5.3.1](https://github.com/OpenTabletDriver/OpenTabletDriver/releases/tag/v0.5.3.1), not 0.4.1.
> 
> [image]

im retarded

X9VoiD, 2021-04-24:
Just to be clear, when does it jump to the top left corner?

naenaebabyy, 2021-04-24:
> Just to be clear, when does it jump to the top left corner?

When it was detected that i was putting the pen near the tablet

X9VoiD, 2021-04-24:
OTD -> Help -> Export Diagnostics, then upload the file here in zipped format

naenaebabyy, 2021-04-24:
Im on the new version now lemme test

naenaebabyy, 2021-04-24:
Fixed old version was problem

closed 2021-04-24 by X9VoiD