OpenTabletDriver/OpenTabletDriver #1047: cth 480 not detected

naenaebabyy, 2021-04-24:
## Description

## System Information:

| Name             | Value |
idk what your meant to put here
| Operating System |
win 10
| Software Version |
latest
| Tablet           |
cth480

X9VoiD, 2021-04-24:
OTD -> Help -> Export Diagnostics, upload file here in zipped format.

naenaebabyy, 2021-04-24:
here it is
[cth480.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6370202/cth480.zip)

X9VoiD, 2021-04-24:
`"The configuration directory 'C:\\Users\\billy\\Desktop\\Configurations' does not exist."`.

Reinstall your OTD properly. If this is from a zip file then you should have extracted everything to a folder of it's own, instead of in `Desktop`. Also, update your plugins.

closed 2021-04-24 by X9VoiD

naenaebabyy, 2021-04-24:
oh