OpenTabletDriver/OpenTabletDriver #1048: Cursor vibrates up and down in absolute mode

Doksha, 2021-04-24:
## Description

When I start the software the cursor vibrates up and down when the pen is near the surface, it seems to be affected by the area.
I saw that there was a similar issue 4 hours ago, but updating to the latest software does not seem to fix the issue, switching to relative mode has similar issues, but instead of actively moving the cursor, it simply causes the cursor to bug around when the pen moves
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Windows 10
| Software Version | 0.5.3.1
| Tablet           | Veikk A15

X9VoiD, 2021-04-24:
Is there any other driver running in the background? And does the vibration look like a cursor ghosting (two cursors)?

Doksha, 2021-04-24:
> Is there any other driver running in the background? And does the vibration look like a cursor ghosting (two cursors)?

I made sure (okay the weather made sure) that there were no other running drivers, I also uninstalled previous drivers so there should be no issues there, and yes, the cursor is ghosting.

Doksha, 2021-04-25:
Fixed the issue, needed to search for two other tablet drivers that were installed, they had weird names so it took a while to find, but that did fix it.

closed 2021-04-25 by X9VoiD