OpenTabletDriver/OpenTabletDriver #1049: Binding handler isn't inside the output pipeline.

adryzz, 2021-04-25:
## Description
With the new change to `IPositionedPipelineElement<T>`, plugins have now access to the `IDeviceReport`s, and can change their type, maybe creating a new type of report that inherits from both `ITabletReport` and `IEraserReport`, essentially adding eraser to tablets that don't support it.

This works with pressure, hover distance and tilt, but not with buttons, pen tip and eraser (essentially anything that is handled by the binding handler)

InfinityGhost, 2021-04-26:
I think wrapping reports in other types isn't a good way about doing that. However, the binding handler *should* be at the end of the pipeline so this is still a worthy change.

InfinityGhost, 2021-06-15:
Resolved in #1098

closed 2021-06-15 by InfinityGhost