OpenTabletDriver/OpenTabletDriver #1066: Add support for VEIKK VK640

TLRS69, 2021-04-28:
# Tablet Information

| Name            | Value |
| --------------- | ----- |
| Manufacturer    | VEIKK
| Product Name    | VEIKK VK640
| Width           | 152.4mm
| Height          | 101.6mm
| Pressure Levels | 8192
| LPI             | 5080

## Diagnostic Information

X9VoiD, 2021-04-28:
OTD -> Help -> Export Diagnostics, upload the file in zipped format

TLRS69, 2021-04-28:
> OTD -> Help -> Export Diagnostics, upload the file in zipped format

TLRS69, 2021-04-28:
[VEIKK VK640.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6391725/VEIKK.VK640.zip)

X9VoiD, 2021-04-28:
Wait until #1063 gets merged

InfinityGhost, 2021-05-03:
Bump, this is ready for a PR

TLRS69, 2021-05-04:
How?

InfinityGhost, 2021-05-04:
Download a [development build](https://github.com/OpenTabletDriver/OpenTabletDriver/actions?query=event%3Apush+is%3Acompleted) of OpenTabletDriver and put the file in this zip in the `Configurations` folder.

[VK640.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6418978/VK640.zip)

LMK if this works, there's still more work to be done even if it does work.

TLRS69, 2021-05-04:
It doesn't work

TLRS69, 2021-05-04:
It detected it at first but once i put my pen on the tablet it doesn't detect it anymore

rigellroy, 2021-06-11:
Sorry for reviving this, but any update on the situation? Thinking of buying another tablet but don't know whether to get the S640 or VK640.

InfinityGhost, 2021-06-13:
> Sorry for reviving this, but any update on the situation? Thinking of buying another tablet but don't know whether to get the S640 or VK640.

No progress, adding new configurations is something that can take quite a bit of trial and error. The VEIKK S640 is a known working tablet if you're still searching for a tablet that OpenTabletDriver supports.

InfinityGhost, 2021-07-19:
Pushed to the backlog until we have a solution for the aux button quirk.

Garbodow, 2021-10-15:
Hey, I was just wondering if there has been any progress on the support on VK640? Had mine for a while now and there really aren't any decent tablet drivers out there that support this model. :(

Mrcubix, 2021-10-15:
An advice for you: Read.

closed 2021-10-17 by InfinityGhost