OpenTabletDriver/OpenTabletDriver #1067: Support for Huion DWH69?

dezgasting, 2021-04-28:
Not sure if it's possible, but I have a Huion DWH69, and OpenTabletDriver doesn't detect it in Linux Mint.

X9VoiD, 2021-04-28:
OTD -> Help -> Export Diagnostics, upload file in zipped format

dezgasting, 2021-04-28:
For some reason it exports a completely empty file.

X9VoiD, 2021-04-28:
Wait until #1042 gets merged.

InfinityGhost, 2021-04-28:
Since this is Linux we really don't need the diagnostics, it can be done manually.

Send the output of `lsusb -v` in a file here. You can automate this by appending ` > file.txt` which will write the output to the file.

dezgasting, 2021-04-29:
[file.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6398300/file.txt)

mzdev-git, 2025-11-12:
the usb port on it is broken sadly, so I don't think so. I can do the driver uninstall thing though