OpenTabletDriver/OpenTabletDriver #1068: Support Request: My Gaomon S56K doesnt work with OTD

dasawar, 2021-04-28:
# Support Request

My Gaomon S56K doesnt work with OTD.
Immediately after I first installed OTD, it did not detect the S56K, but after edit the configuration, it detect my S56K.
However, it does not work in the area I set on the OTD so the S56K is not recognized correctly and Windows INK enabled.
no chance to work my S56K with OTD?
## What I've already tried

https://www.youtube.com/watch?v=35-Jz6_TZOc
OTD Installation and FAQ page
WinUSB cleanup script
## Diagnostic Information

[Diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6391615/Diagnostics.txt)
[S56K.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6391616/S56K.txt)

X9VoiD, 2021-04-28:
OTD -> Tablet -> Device string reader

VID: 9580
PID: 110

Leave Index empty, then press Dump all. Upload the generated dump here.

dasawar, 2021-04-28:
[stringdump.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6394519/stringdump.txt)

X9VoiD, 2021-04-28:
Are you sure that this is a Gaomon S56K?

X9VoiD, 2021-04-28:
I reviewed the diagnostics and by the looks of it, the device requires a WinUSB filter driver for it to work with a driver other than the PnP Windows Ink.

Wait until #475 is resolved.

dasawar, 2021-04-28:
I understand. I will wait for that driver. Thank you for your reply.

EikelFlo, 2021-07-30:
Yeah, I have the same issue and my dump says Huion Tablet LuoYong70558 will just wait until its solved whenever the backlog is cleared

Mrcubix, 2021-07-30:
> Yeah, I have the same issue and my dump says Huion Tablet LuoYong70558 will just wait until its solved whenever the backlog is cleared

Open your own issue, this probably has nothing to do with your

Mrcubix, 2021-09-09:
Join the [discord server](https://discord.gg/P3GYdUS) when you have time, WinUSB support has been added in pre-release and it will be faster to get it compatible if you join

dasawar, 2021-09-10:
Thank you for working for this issue. I've already got CTL672, but I still have Gaomon S56K so I will give it a try.

Mrcubix, 2021-09-10:
> Thank you for working for this issue. I've already got CTL672, but I still have Gaomon S56K so I will give it a try.

it's mainly for the sake of growing OTD's compatibility list, join the help-windows channel there so we can start

gonX, 2021-12-06:
Closing as stale support issue. Gaomon S56K is now supported but requires WinUSB setup on Windows. Please visit our Discord for help.

closed 2021-12-06 by gonX