OpenTabletDriver/OpenTabletDriver #1071: Cursor Clipping 1px to top/left in absolute mode (1px bottom/right in winink abs. mode)

OniCado, 2021-04-29:
## Description
When setting the desktop area to a monitor, the cursor will clip to the monitor next to it (top, left in normal absolue mode, bottom, right in winink abs. mode). This does not happen w/ the XP-PEN G640 when configured as full area, since it does not report any pen position outside its usable area.

## System Information:
| Name            | Value                                                                                  |
| -------------- | -------------------------------------------------------------- |
| OS                 | Windows 10 Pro N Build 19042, Win10 Enterprise 19042 |
| OTD Version | v0.5.3, v0.5.2.3, v0.5.3.1                                                       |
| Tablets          | CTx-x80, CTL-672, XP-PEN Star G640 rev A*                      |

InfinityGhost, 2021-04-29:
Seems like we need to offset the output on the WindowsAbsolutePointer by 1px X & Y in that case. This doesn't appear to occur on Linux.

doubledge-spec, 2021-05-03:
I am an EndeavourOS, an arch linux flavour. My cursor by default goes two pixels into my second monitor unless i offset it in the driver

doubledge-spec, 2021-05-03:
I have not tried other monitor directions

gonX, 2021-12-25:
Can confirm this happens on Linux btw but it seems to be 1 pixel rather than 2. I'm changing milestone to 0.7.x

gonX, 2026-02-26:
Moving this to v0.6.8 milestone as resolving this in a clean manner does not seem clear.

I believe this is caused by a combination of some units being zero-inclusive (like tablet positions) while others aren't (monitor pixels, yet most output modes are 0-inclusive), so the final value truncation to `int` can cause some weird side effects (except macOS which directly takes the positional values as `double`s)

For example, Windows' [MOUSEINPUT struct docs](https://learn.microsoft.com/en-us/windows/win32/api/winuser/ns-winuser-mouseinput#remarks) says the following:
> If MOUSEEVENTF_ABSOLUTE value is specified, dx and dy contain normalized absolute coordinates between 0 and 65,535. The event procedure maps these coordinates onto the display surface. Coordinate (0,0) maps onto the upper-left corner of the display surface; coordinate (65535,65535) maps onto the lower-right corner. 

Normally this works fine, but suppose a monitor setup of 2x 1920px wide, you have 3840 individual pixels per row to place your cursor on but OTD works in 3841 units there, which is likely then exacerbated by casting the positional `float` values to `int`:

https://github.com/OpenTabletDriver/OpenTabletDriver/blob/3606ea1e005121eb8c79d7bd8cb3dd13134bd3a4/OpenTabletDriver.Desktop/Interop/Input/Absolute/WindowsAbsolutePointer.cs#L18-L19

Again, it's not clear to me which part is going wrong where, so more research is needed.