OpenTabletDriver/OpenTabletDriver #1084: detects the cth460 but nothing works

awesomeboxlord, 2021-05-03:
## Description

so i try to use the drivers but all i get is it saying it sees the 460 and nothing works, the leds on the tablet do a better job at telling that im doing something, the input info is only detecting the touch input and that is pretty much it, no buttons work the pen doesnt work, the touch technically is detected but it doesnt seem to respond much farther than that
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System windows 10 |
| Software Version v0.5.3.1 |
| Tablet cth-460 |

Mrcubix, 2021-05-03:
uninstall wacom drivers if installed

awesomeboxlord, 2021-05-03:
the problem is i already have it uninstalled, i have hawku drivers but i made sure to close it when testing, and there was a test where i didnt even open it on startup and only opened up the open tablet driver

Ys0505, 2021-05-06:
I have the same problem on my huion hs64 but I am using mac

InfinityGhost, 2021-05-12:
> the problem is i already have it uninstalled, i have hawku drivers but i made sure to close it when testing, and there was a test where i didnt even open it on startup and only opened up the open tablet driver

If they are installed *at all*, it loads a kernel driver at boot that triggers the device endpoints to have different identifiers.

> I have the same problem on my huion hs64 but I am using mac

Unrelated, open a separate issue.

InfinityGhost, 2021-07-18:
Stale support request, no update in months. Assuming resolved.

closed 2021-07-18 by InfinityGhost