OpenTabletDriver/OpenTabletDriver #1090: Support Request: cursor stuck in game OSU using OTD

cr1smc, 2021-05-06:
I'm trying to play osu with G640 usind OTD, in desktop or any other app, the cursor is moving accordingly, but in osu, after debugging for an hour it won't work, i tried everything that has to be tried on google.
if you have any ideas what could make this happen please let me know

gonX, 2021-05-06:
Make sure raw input is disabled, and make sure you're not running osu! as admin

cr1smc, 2021-05-06:
Oh, i'm actually running as admin, uh, my game is kinda broken it only starts if i start as admin. will try to fix the admin thing first
then i'll come back with an answer.

cr1smc, 2021-05-06:
lifesaver my man, was the admin rights that messed up the driver. Cheers

closed 2021-05-06 by X9VoiD