OpenTabletDriver/OpenTabletDriver #1091: Plugins fail to load on Manjaro.

nekkan, 2021-05-07:
## Description
Every plugin I try to install results in a warning message in the console.
[image]

## System Information:

| Name             | Value |
| ---------------- | ----- | 
| Operating System | Manjaro 21.0.4
| Software Version | v0.5.3.1-r188.gf2ec9fd4
| Tablet           | Gaomon S620

InfinityGhost, 2021-05-07:
We have changed a large majority of the plugin library, breaking basically all v0.5.3 plugins. The PKGBUILD can be modified to directly fetch the v0.5.3.1 release so you won't be building on latest master, if needed.

closed 2021-05-07 by InfinityGhost

nekkan, 2021-05-07:
How can I modify the PKGBUILD?