OpenTabletDriver/OpenTabletDriver #1094: Support Request: XP PEN G430S When hovering over the tablet it gets undetected?

rudyon, 2021-05-08:
# Support Request

The tablet is detected after plugging it in. However when I go to use the tablet. As soon as my pen hovers over. The driver stops detecting it.

## What I've already tried

- Plugging into a different USB port.
- Rebooting.
- Reinstalling several times.

## Diagnostic Information

[Privatebin](https://privatebin.net/?9007a439753b7e7e#694Be98XbMJ53GfGRdReNpFgvvDvzXU3xz8wZc4KPFiq)

Jummit, 2021-05-09:
I'm having a simillar issue with a Wacom tablet.
```json
[NullReferenceException:Error]	Object reference not set to an instance of an object.
   at OpenTabletDriver.Desktop.DesktopDriver.HandleReport(IDeviceReport report) in /home/jummit/.cache/yay/opentabletdriver-git/src/OpenTabletDriver/OpenTabletDriver.Desktop/DesktopDriver.cs:line 10
   at OpenTabletDriver.Devices.DeviceReader`1.Main() in /home/jummit/.cache/yay/opentabletdriver-git/src/OpenTabletDriver/OpenTabletDriver/Devices/DeviceReader.cs:line 99
```

Faithy3, 2021-05-25:
Just encountered this issue. Arch Linux kernel 5.12.5-arch1-1, software version v0.5.3.1-r192.g03d1d05e. The tablet's a CTH-680

ixhbinphoenix, 2021-06-10:
Have the same issue. Manjaro kernel 5.10.41-1-MANJARO OpenTabletDriver v0.5.3.1-r221.g96516285. Tablet is a Wacom CTL-472

InfinityGhost, 2021-06-13:
Odds are this is out of date plugins, can't reproduce in any scenario.

InfinityGhost, 2021-07-18:
Stale support request, assuming resolved.

closed 2021-07-18 by InfinityGhost