OpenTabletDriver/OpenTabletDriver #1096: Support Request:

ElboyYT, 2021-05-11:
# Support Request
 OTD doesn't detect my Tablet (Gaomon s620)
 My Tablet is pluged in and works fine, but it won't detect in OTD.

## What I've already tried
 I've tried repluging my Tablet, re open the OTD software

## Diagnostic Information

{
  "App Version": "OpenTabletDriver v0.5.3.1",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.19042.0",
    "VersionString": "Microsoft Windows NT 10.0.19042.0"
  },
  "Environment Variables": {
    "PSModulePath": "C:\\Program Files\\WindowsPowerShell\\Modules;C:\\WINDOWS\\system32\\WindowsPowerShell\\v1.0\\Modules",
    "LOCALAPPDATA": "C:\\Users\\elboy\\AppData\\Local",
    "TEMP": "C:\\Users\\elboy\\AppData\\Local\\Temp",
    "PUBLIC": "C:\\Users\\Public",
    "SystemRoot": "C:\\WINDOWS",
    "ALLUSERSPROFILE": "C:\\ProgramData",
    "TMP": "C:\\Users\\elboy\\AppData\\Local\\Temp",
    "OneDriveConsumer": "C:\\Users\\elboy\\OneDrive",
    "CommonProgramFiles": "C:\\Program Files\\Common Files",
    "OS": "Windows_NT",
    "windir": "C:\\WINDOWS",
    "ProgramData": "C:\\ProgramData",
    "ComSpec": "C:\\WINDOWS\\system32\\cmd.exe",
    "OneDrive": "C:\\Users\\elboy\\OneDrive",
    "USERNAME": "Elboy",
    "NUMBER_OF_PROCESSORS": "16",
    "CommonProgramW6432": "C:\\Program Files\\Common Files",
    "PATHEXT": ".COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC",
    "ProgramFiles(x86)": "C:\\Program Files (x86)",
    "COMPUTERNAME": "PC-ELYES",
    "LOGONSERVER": "\\\\PC-ELYES",
    "CommonProgramFiles(x86)": "C:\\Program Files (x86)\\Common Files",
    "USERDOMAIN_ROAMINGPROFILE": "PC-ELYES",
    "ProgramFiles": "C:\\Program Files",
    "USERDOMAIN": "PC-ELYES",
    "PROCESSOR_LEVEL": "23",
    "APPDATA": "C:\\Users\\elboy\\AppData\\Roaming",
    "PROCESSOR_REVISION": "0802",
    "HOMEPATH": "\\Users\\elboy",
    "SystemDrive": "C:",
    "PROCESSOR_ARCHITECTURE": "AMD64",
    "HOMEDRIVE": "C:",
    "USERPROFILE": "C:\\Users\\elboy",
    "Path": "C:\\Program Files (x86)\\Razer Chroma SDK\\bin;C:\\Program Files\\Razer Chroma SDK\\bin;C:\\Program Files (x86)\\Razer\\ChromaBroadcast\\bin;C:\\Program Files\\Razer\\ChromaBroadcast\\bin;C:\\Program Files (x86)\\Common Files\\Oracle\\Java\\javapath;C:\\Windows\\system32;C:\\Windows;C:\\Windows\\System32\\Wbem;C:\\Windows\\System32\\WindowsPowerShell\\v1.0\\;C:\\Windows\\System32\\OpenSSH\\;C:\\Windows\\system32\\config\\systemprofile\\AppData\\Local\\Microsoft\\WindowsApps;C:\\Users\\elboy\\AppData\\Local\\Microsoft\\WindowsApps;C:\\WINDOWS\\system32;C:\\WINDOWS;C:\\WINDOWS\\System32\\Wbem;C:\\WINDOWS\\System32\\WindowsPowerShell\\v1.0\\;C:\\WINDOWS\\System32\\OpenSSH\\;C:\\Program Files\\dotnet\\;C:\\Users\\elboy\\AppData\\Local\\Microsoft\\WindowsApps;",
    "PROCESSOR_IDENTIFIER": "AMD64 Family 23 Model 8 Stepping 2, AuthenticAMD",
    "DriverData": "C:\\Windows\\System32\\Drivers\\DriverData",
    "ProgramW6432": "C:\\Program Files",
    "SESSIONNAME": "Console"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\hid#vid_1532&pid_0046&mi_01&col01#8&2e7bbaf9&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd",
      "VendorID": 5426,
      "ProductID": 70,
      "CanOpen": false,
      "ReleaseNumberBcd": 512,
      "ReleaseNumber": "2.0",
      "ProductVersion": 512,
      "Manufacturer": "Razer",
      "ProductName": "Razer Mamba Tournament Edition",
      "SerialNumber": "",
      "MaxInputReportLength": 16,
[... 4884 lines cut ...]
      "Time": "2021-05-11T19:09:48.3207085+02:00",
      "Group": "Detect",
      "Message": "Searching for tablet 'XP-Pen Star G640S'",
      "Level": 1
    },
    {
      "Time": "2021-05-11T19:09:48.3207161+02:00",
      "Group": "Detect",
      "Message": "Searching for tablet 'XP-Pen Star G960'",
      "Level": 1
    },
    {
      "Time": "2021-05-11T19:09:48.3207237+02:00",
      "Group": "Detect",
      "Message": "Searching for tablet 'XP-Pen Star G960S Plus'",
      "Level": 1
    },
    {
      "Time": "2021-05-11T19:09:48.3207314+02:00",
      "Group": "Detect",
      "Message": "No tablet found.",
      "Level": 1
    }
  ]
}

ElboyYT, 2021-05-11:
I also still have the Gaomon Driver, but not active.

X9VoiD, 2021-05-11:
Uninstall Gaomon Driver, then replug tablet, if that did not help then use [WinUSBCleanupScript](https://github.com/X9VoiD/WinUSBCleanup/releases/tag/v1.0). Will close since this is a common issue with known fix.

closed 2021-05-11 by X9VoiD