OpenTabletDriver/OpenTabletDriver #1097: Fedora install : need help

QuintusNX, 2021-05-11:
# Support Request

Hello, I have the xp pen star g640s and i run fedora 34, si I've tried to install it.
But because I'm still a "beginer" at linux, idk how to compile this code and found nothing understandable online out there

## What I've already tried

What I tried is to translate the deb package to rpm format with alien.
The translation worked but the installation crahes

## Diagnostic Information

sudo dnf install '/root/OpenTabletDriver-0.5.3.1-2.x86_64.rpm'

[Yes, install the package] : 

Downloading Packages:
Running transaction check
Transaction check succeeded.
Running transaction test
Error: Transaction test error:
  file /usr/bin from install of OpenTabletDriver-0.5.3.1-2.x86_64 conflicts with file from package filesystem-3.14-5.fc34.x86_64
  file /usr/lib from install of OpenTabletDriver-0.5.3.1-2.x86_64 conflicts with file from package filesystem-3.14-5.fc34.x86_64

So, at this state, idk what to do, and how to install or compile the package.
Can anyone help me ?

InfinityGhost, 2021-05-12:
FWIW we plan on adding a rpm package in v0.6 so this won't be a problem then. For now we'll leave this open to track the progress.

diaryunderneathatree, 2021-06-25:
Looking forward to an rpm package too, on Fedora 34 , it'll make my workflow much more simple!

cxr6548, 2021-07-07:
Dropping by the third the request!

InfinityGhost, 2021-07-14:
https://github.com/OpenTabletDriver/OpenTabletDriver.Packaging/pull/3

InfinityGhost, 2021-07-17:
We haven't added CI for the RPM package, however it does work. We will be including it in future releases.

closed 2021-07-17 by InfinityGhost

glow12121, 2023-06-11:
Please include a Fedora package.