OpenTabletDriver/OpenTabletDriver #1100: Add Lock to display

Naozumi520, 2021-05-14:
## Description
So I have got a second monitor, but after that my tablet area was weird, so I have found that I need to use the "Set to display" option, but after that my tablet still able to control the second monitor and the aspect ratio of the tablet area is wrong.
Would be nice if there could be an option to just use the single screen then we don't need to edit the current area.

X9VoiD, 2021-05-14:
Tablet Area Editor -> Right Click -> Ignore input outside area

X9VoiD, 2021-05-14:
For the S640 measurements open up a new issue.

Open up Tablet -> Tablet Debugger and post there the highest values you can get for Position

closed 2021-05-14 by X9VoiD

Naozumi520, 2021-05-14:
Hmmmm Idk why but yeah I have choose Ignore input outside area, but I can still move the cursor to the other screen.