OpenTabletDriver/OpenTabletDriver #1102: Key bindings don't work

OvoJoaca, 2021-05-14:
## Description

Binding a pen button to a keyboard key doesn't work.
For example I set my Pen Button 1 to Escape and after I save and apply the settings and try to press the button on the pen, PC doesn't seem to recognize that I'm pressing anything.
Also the Mouse bindings work.
https://imgur.com/a/iQB3nIz
## System Information:

| Name: ovo | Value |
| ---------------- | ----- |
| Operating System: Manjaro Linux x86_64 |
| Software Version: v0.5.3.1-r188.gf2ec9fd4 |
| Tablet: Wacom CTL-472 |

X9VoiD, 2021-05-24:
Have you already tried rebinding it? Can't reproduce on Arch.

OvoJoaca, 2021-05-24:
Yes! I tried. It worked the first time I installed it (a few months ago) then I updated my kernel (to linux512) and noticed that it wouldn't start due to something about gtk so I tried reinstalling it from aur and it worked, but not the keybindings. I had the same issue with kcm-wacomtablet where the mouse keybinds would work, but the keyboard keybinds wont.

ashblk, 2021-05-28:
This is likely caused by a bug in KeyBindings.Press and KeyBindings.Release, where the string null-or-empty check was not negated, causing it to return false for valid non-empty strings. 
I made a pull request addressing it: #1127

InfinityGhost, 2021-06-09:
Resolved in #1127

closed 2021-06-09 by InfinityGhost