OpenTabletDriver/OpenTabletDriver #1105: Not detecting my tablet (Huion H640P)

drpure, 2021-05-16:
## Description

Hi, i just recently switched form using hawku to opentabletdriver because someone told me that this driver gives lower latency. sorry for bothering you guys
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Windows 10 |
|5.3.1 |
| Huion 640P        |

X9VoiD, 2021-05-24:
OTD -> Help -> Export Diagnostics, then upload the diagnostics as a zipped file. Will close the issue for now until further information is provided.

closed 2021-05-24 by X9VoiD

drpure, 2021-05-31:
[H640P.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6571785/H640P.zip)

drpure, 2021-05-31:
when I look at console, it says "tablet found", but on the top of the application it says "no tablet detected"

Mrcubix, 2021-05-31:
[image]
https://github.com/X9VoiD/WinUSBCleanup/releases/tag/v1.0

drpure, 2021-05-31:
still doesn't work

Mrcubix, 2021-05-31:
export diag again

drpure, 2021-06-01:
[H640P (5).zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6573170/H640P.5.zip)

Mrcubix, 2021-06-01:
the previously linked cleanup script wasn't launched or it didn't work for a reason or another, run it as admin just to make sure and it should work

drpure, 2021-06-01:
it worked, ty!