OpenTabletDriver/OpenTabletDriver #1106: Support Request:

Andreas-Atomphi, 2021-05-17:
# No tablet detected

I use a BamBoo 4x5 on a Linux Ubuntu 20.04, i don't know why, but xsetwacom detect that is connected but OpenTabletDrive not, i don't know if that tablet has no support or if is a bug.

## I don't know what to try.
Every solution is for a problem with bugs, and are all for Windows, so i don't know what i have to do, the Wacom drivers are all up to date.

## Diagnostic Information

Debug infos:
[Plugin:Debug]	Loading plugin 'Precision Control'
[Detect:Info]	Searching for tablet 'Wacom PTH-451'
[Detect:Info]	Searching for tablet 'Wacom CTL-671'
[Detect:Info]	Searching for tablet 'Wacom CTH-490'
[Detect:Info]	Searching for tablet 'Wacom CTL-4100 Bluetooth'
[Detect:Info]	Searching for tablet 'Wacom CTH-480'
[Detect:Info]	Searching for tablet 'Wacom CTL-6100'
[Detect:Info]	Searching for tablet 'Wacom PTH-660'
[Detect:Info]	Searching for tablet 'Wacom CTH-670'
[Detect:Info]	Searching for tablet 'Wacom CTL-4100WL'
[Detect:Info]	Searching for tablet 'Wacom MTE-450'
[Detect:Info]	Searching for tablet 'Wacom CTL-490'
[Detect:Info]	Searching for tablet 'Wacom CTL-472'
[Detect:Info]	Searching for tablet 'Wacom PTH-651'
[Detect:Info]	Searching for tablet 'Wacom FT-0405-U'
[Detect:Info]	Searching for tablet 'Wacom CTL-690'
[Detect:Info]	Searching for tablet 'Wacom PTH-860'
[Detect:Info]	Searching for tablet 'Wacom XD-0608-U'
[Detect:Info]	Searching for tablet 'Wacom CTL-672'
[Detect:Info]	Searching for tablet 'Wacom CTL-4100'
[Detect:Info]	Searching for tablet 'Wacom CTH-680'
[Detect:Info]	Searching for tablet 'Wacom CTL-480'
[Detect:Info]	Searching for tablet 'Wacom CTL-460'
[Detect:Info]	Searching for tablet 'Wacom PTK-1240'
[Detect:Info]	Searching for tablet 'Wacom CTH-470'
[Detect:Info]	Searching for tablet 'Wacom CTE-430'
[Detect:Info]	Searching for tablet 'Wacom CTE-650'
[Detect:Info]	Searching for tablet 'Wacom PTK-540WL'
[Detect:Info]	Searching for tablet 'Wacom CTE-440'
[Detect:Info]	Searching for tablet 'Wacom CTH-460'
[Detect:Info]	Searching for tablet 'Wacom PTH-850'
[Detect:Info]	Searching for tablet 'Wacom CTH-690'
[Detect:Info]	Searching for tablet 'Wacom PTZ-930'
[Detect:Info]	Searching for tablet 'Wacom CTL-680'
[Detect:Info]	Searching for tablet 'Wacom CTE-460'
[Detect:Info]	Searching for tablet 'Wacom CTL-470'
[Detect:Info]	Searching for tablet 'Wacom PTK-440'
[Detect:Info]	Searching for tablet 'Wacom CTL-471'
[Detect:Info]	Searching for tablet 'Wacom PTH-660 Bluetooth'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 01'
[Detect:Info]	Searching for tablet 'XP-Pen G540'
[Detect:Info]	Searching for tablet 'XP-Pen G960 v2'
[Detect:Info]	Searching for tablet 'XP-Pen Star 03'
[Detect:Info]	Searching for tablet 'XP-Pen Artist Pro 12'
[Detect:Info]	Searching for tablet 'XP-Pen Deco Mini7'
[Detect:Info]	Searching for tablet 'XP-Pen G640s'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 02'
[Detect:Info]	Searching for tablet 'XP-Pen G430S'
[Detect:Info]	Searching for tablet 'XP-Pen Star G960S Plus'
[Detect:Info]	Searching for tablet 'XP-Pen G640 v2'
[Detect:Info]	Searching for tablet 'XP-Pen G430'
[Detect:Info]	Searching for tablet 'XP-Pen G540 Pro'
[Detect:Info]	Searching for tablet 'XP-Pen Star 06C'
[Detect:Info]	Searching for tablet 'XP-Pen G640'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 01 v2'
[Detect:Info]	Searching for tablet 'XP-Pen Star 06'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 03'
[Detect:Info]	Searching for tablet 'XP-Pen Deco Pro Small'
[Detect:Info]	Searching for tablet 'Huion New 1060 Plus'
[Detect:Info]	Searching for tablet 'Huion Inspiroy Q11K V2'
[... 487 lines cut ...]
[Detect:Info]	Searching for tablet 'Huion H1060P'
[Detect:Info]	Searching for tablet 'Huion Kamvas 16'
[Detect:Info]	Searching for tablet 'Huion HS64'
[Detect:Info]	Searching for tablet 'Huion HS611'
[Detect:Info]	Searching for tablet 'Huion HC16'
[Detect:Info]	Searching for tablet 'Huion osu!tablet'
[Detect:Info]	Searching for tablet 'Huion H610 Pro V2'
[Detect:Info]	Searching for tablet 'Huion H640P'
[Detect:Info]	Searching for tablet 'Huion G10T'
[Detect:Info]	Searching for tablet 'Huion Q620M'
[Detect:Info]	Searching for tablet 'Huion 1060 Plus'
[Detect:Info]	Searching for tablet 'Gaomon S620'
[Detect:Info]	Searching for tablet 'Gaomon S56K'
[Detect:Info]	Searching for tablet 'Gaomon M106K'
[Detect:Info]	Searching for tablet 'Gaomon M106K Pro'
[Detect:Info]	Searching for tablet 'Gaomon 1060 Pro'
[Detect:Info]	Searching for tablet 'Gaomon M10K'
[Detect:Info]	Searching for tablet 'VEIKK S640'
[Detect:Info]	Searching for tablet 'VEIKK A30'
[Detect:Info]	Searching for tablet 'VEIKK A50'
[Detect:Info]	Searching for tablet 'VEIKK A15'
[Detect:Info]	Searching for tablet 'VEIKK A15 Pro'
[Detect:Info]	No tablet found.

No errors and no warnings.

TheFantasticLoki, 2021-05-24:
@InfinityGhost  Having same issue no tablet detected. Restarted Computer, restarted drivers afterwords, still nothing, I have the proper configs as they were just working a few days ago before I left for a trip. Running Windows 10 Pro, Default Microsoft Drivers for tablet disabled to stop drivers from fighting, Using a XP-PEN Star 06C. Will Attach Diagnostics and Configs:

Diagnostics: `{
  "App Version": "OpenTabletDriver v0.5.3.1",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.19042.0",
    "VersionString": "Microsoft Windows NT 10.0.19042.0"
  },
  "Environment Variables": {
    "LOGONSERVER": "\\\\DESKTOP-L0K1",
    "VBOX_MSI_INSTALL_PATH": "C:\\Program Files\\Oracle\\VirtualBox\\",
    "DriverData": "C:\\Windows\\System32\\Drivers\\DriverData",
    "COMPUTERNAME": "DESKTOP-L0K1",
    "TA_GAME_PATH": "F:\\Games\\SteamLibrary\\steamapps\\common\\Call of Duty Black Ops III\\",
    "HOMEPATH": "\\Users\\astey",
    "TEMP": "C:\\Users\\astey\\AppData\\Local\\Temp",
    "SNF": "C:\\ProgramData\\Snorlers\\snp.sc",
    "ProgramW6432": "C:\\Program Files",
    "SystemDrive": "C:",
    "OneDrive": "C:\\Users\\astey\\OneDrive",
    "USERPROFILE": "C:\\Users\\astey",
    "OculusBase": "C:\\Program Files\\Oculus\\",
    "PSModulePath": "C:\\Program Files\\WindowsPowerShell\\Modules;C:\\WINDOWS\\system32\\WindowsPowerShell\\v1.0\\Modules",
    "NUMBER_OF_PROCESSORS": "16",
    "ProgramFiles(x86)": "C:\\Program Files (x86)",
    "Path": "C:\\Program Files (x86)\\VMware\\VMware Workstation\\bin\\;C:\\ProgramData\\Oracle\\Java\\javapath;C:\\Program Files (x86)\\Common Files\\Intel\\Shared Libraries\\redist\\intel64\\compiler;C:\\Program Files\\Oculus\\Support\\oculus-runtime;C:\\Program Files\\Java\\jre1.8.0_281\\bin;C:\\Program Files\\dotnet\\;C:\\Program Files\\nodejs\\;C:\\Program Files\\Geth;C:\\Program Files (x86)\\dotnet\\;C:\\Program Files (x86)\\NVIDIA Corporation\\PhysX\\Common;C:\\Program Files\\Process Lasso\\;C:\\WINDOWS\\system32;C:\\WINDOWS;C:\\WINDOWS\\System32\\Wbem;C:\\WINDOWS\\System32\\WindowsPowerShell\\v1.0\\;C:\\WINDOWS\\System32\\OpenSSH\\;C:\\Program Files\\Go\\bin;C:\\Program Files\\Git\\cmd;C:\\Users\\astey\\AppData\\Local\\Programs\\Python\\Python39\\Scripts\\;C:\\Users\\astey\\AppData\\Local\\Programs\\Python\\Python39\\;C:\\Program Files\\Java\\jre1.8.0_261\\bin;C:\\Users\\astey\\AppData\\Roaming\\npm;C:\\Users\\astey\\AppData\\Local\\GitHubDesktop\\bin;C:\\Users\\astey\\AppData\\Local\\Programs\\Microsoft VS Code\\bin;C:\\Windows\\system32;;C:\\Users\\astey\\AppData\\Local\\Microsoft\\WindowsApps;C:\\Users\\astey\\go\\bin;C:\\Users\\astey\\.dotnet\\tools",
    "PUBLIC": "C:\\Users\\Public",
    "TA_TOOLS_PATH": "F:\\Games\\SteamLibrary\\steamapps\\common\\Call of Duty Black Ops III\\",
    "INTEL_DEV_REDIST": "C:\\Program Files (x86)\\Common Files\\Intel\\Shared Libraries\\",
    "ProgramData": "C:\\ProgramData",
    "PROCESSOR_LEVEL": "23",
    "TMP": "C:\\Users\\astey\\AppData\\Local\\Temp",
    "SNP": "https://%66%65%65%64.%68%65%6C%70%65%72%62%61%72.%63%6F%6D?publisher=APSFFreeSS&co=US&userid=e6aa3523-5b11-50f5-799e-8bcde9144128&searchtype=sc&installDate=05/04/2020&barcodeid=53724003&channelid=3&av=windows",
    "LOCALAPPDATA": "C:\\Users\\astey\\AppData\\Local",
    "PATHEXT": ".COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC",
    "HOMEDRIVE": "C:",
    "OneDriveConsumer": "C:\\Users\\astey\\OneDrive",
    "CommonProgramFiles": "C:\\Program Files\\Common Files",
    "ProgramFiles": "C:\\Program Files",
    "PROCESSOR_ARCHITECTURE": "AMD64",
    "ETHEREUM_SOCKET": "\\\\.\\pipe\\geth.ipc",
    "GOPATH": "C:\\Users\\astey\\go",
    "CommonProgramW6432": "C:\\Program Files\\Common Files",
    "ComSpec": "C:\\WINDOWS\\system32\\cmd.exe",
    "OS": "Windows_NT",
    "FPS_BROWSER_APP_PROFILE_STRING": "Internet Explorer",
    "CommonProgramFiles(x86)": "C:\\Program Files (x86)\\Common Files",
    "USERDOMAIN_ROAMINGPROFILE": "DESKTOP-L0K1",
    "CLASSPATH": "C:\\Program Files\\Java\\jre1.8.0_241\\bin",
    "USERDOMAIN": "DESKTOP-L0K1",
    "SESSIONNAME": "Console",
    "MIC_LD_LIBRARY_PATH": "C:\\Program Files (x86)\\Common Files\\Intel\\Shared Libraries\\compiler\\lib\\mic",
    "FPS_BROWSER_USER_PROFILE_STRING": "Default",
    "VBOX_HWVIRTEX_IGNORE_SVM_IN_USE": "1",
    "APPDATA": "C:\\Users\\astey\\AppData\\Roaming",
    "USERNAME": "astey",
    "windir": "C:\\WINDOWS",
    "PROCESSOR_IDENTIFIER": "AMD64 Family 23 Model 8 Stepping 2, AuthenticAMD",
    "TA_LOCAL_ASSET_CACHE": "F:\\Games\\SteamLibrary\\steamapps\\common\\Call of Duty Black Ops III\\share\\assetconvert\\",
    "PROCESSOR_REVISION": "0802",
    "ALLUSERSPROFILE": "C:\\ProgramData",
    "DXSDK_DIR": "C:\\Program Files (x86)\\Microsoft DirectX SDK (June 2010)\\",
    "SystemRoot": "C:\\WINDOWS"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\hid#vid_28bd&pid_0062&mi_00&col01#8&2212d316&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "VendorID": 10429,
[... 684 lines cut ...]
      "MaxX": 50800.0,
      "MaxY": 30480.0,
      "MaxPressure": 8191,
      "ActiveReportID": {
        "Start": 80,
        "StartInclusive": true,
        "End": 96,
        "EndInclusive": false
      },
      "VendorID": 10429,
      "ProductID": 98,
      "InputReportLength": 8,
      "OutputReportLength": 8,
      "ReportParser": "OpenTabletDriver.Vendors.XP_Pen.XP_PenReportParser",
      "FeatureInitReport": null,
      "OutputInitReport": "ArAE",
      "DeviceStrings": {},
      "InitializationStrings": []
    }
  ],
  "AuxilaryDeviceIdentifiers": [],
  "Attributes": {
    "libinputoverride": "1"
  }
}`

X9VoiD, 2021-05-24:
@Andreas-Atomphi the log you posted is not the diagnostic info we needed. Do OTD -> Help -> Export Diagnostics then upload it as a zipped file.

closed 2021-05-24 by X9VoiD