OpenTabletDriver/OpenTabletDriver #1111: Opentabletdriver broken on Arch Linux

kerostampcrab, 2021-05-20:
## Description
If i put my pen on my tablet, it will undetect my tablet, so i would have to use Ctrl + D to redetect it but of course it will undetect once i place it somewhere near my tablet. Also if i install any plugins, it will say "[Plugin:Warning]	Plugin 'Kuuube's_CHATTER_EXTERMINATOR, Version=1.0.0.0' can't be loaded and is likely out of date." in the console. Also there would be an interpolators tab on my laptop (running PopOS!), which there isn't here. I also tried in Manjaro, but it's the same result.

## System Information:
| Operating System | Arch Linux
| Software Version | v0.5.3.1-r188.gf2ec9fd4
| Tablet           | Wacom CTL-490
[logcopiedfromconsole.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6517375/logcopiedfromconsole.txt)

dg0, 2021-05-20:
> v0.5.3.1-r188.gf2ec9fd4
There's your answer, because of some core changes all plugins are broken on latest master. If you want to use plugins/interpolators you have to modify the PKGBUILD to fetch from an older version.

X9VoiD, 2021-05-20:
If you want to use both OTD master and the plugins then build the master branch of the plugins.

closed 2021-05-20 by X9VoiD