OpenTabletDriver/OpenTabletDriver #1120: Exception on Win shutdown

C0D3-M4513R, 2021-05-24:
## Description

I was shutting down my pc, when this appeared:
[image]

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Win 10, version 21H1, build 19043.985
| Software Version | v.0.5.3.1
| Tablet           | MTE-450

X9VoiD, 2021-05-24:
Please provide a stack trace or log, we can't do anything with just the screenshot. Will close for now until further information is provided.

closed 2021-05-24 by X9VoiD

InfinityGhost, 2021-05-24:
This appears to either be a .NET bug or a WPF bug.

C0D3-M4513R, 2021-05-24:
> Please provide a stack trace or log, we can't do anything with just the screenshot. Will close for now until further information is provided.

Where are the logs? Do I need to restart the app, to get to the logs? How do I take a stacktrace? The Window is still open btw.

Mrcubix, 2021-05-24:
One way you can get the stacktrace is by checking the event viewer > windows log > application for .NET Runtime errors related to OTD, from there, copy paste the content of that stacktrace contained in the text box in the middle.
[image]

C0D3-M4513R, 2021-05-24:
I do not have a stacktrace available. I am finding some errors, that were one to many months before, from other apps though.
I am writing this off as a freak accident, and will just not worry about it.
Some Computer wierdness, that probably occurred because of my leniency, to not reboot my computer, when installing/updating stuff?
Don't worry about this!