OpenTabletDriver/OpenTabletDriver #1124: Plugins should not have access to reports filtered out by parsers

Kuuuube, 2021-05-27:
## Description

Currently it is possible for plugins to access reports that are filtered out on the parser side. Reports filtered out by configs are not sent through plugins and config side filtering works as it should. But having parser filtering ignored by plugins can lead to messed up output from plugins that track reports over time. For example: [Tablet Debounce](https://github.com/Kuuuube/Tablet_Debounce) ends up sending wrong pressure values over valid reports because it records bad pressure values from invalid reports that are filtered parser side.
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Software Version | 03d1d05

InfinityGhost, 2021-07-18:
Is this issue still valid with pipeline elements?

Kuuuube, 2021-07-18:
This issue only came up because of pipeline. I believe it is still an issue but the parser changes may have made it not matter anymore. I don't have a good way to test it currently though.

InfinityGhost, 2021-07-18:
Looks like the only fix to this would to basically implement a "disposed" property of sorts to determine if a report should be ignored. 
Not sure on a good fix for this.

gonX, 2021-12-06:
This sounds more like a parser issue that should be handled on a case by case basis.

Kuuuube, 2022-01-12:
This doesn't matter anymore.

jamesbt365, 2022-03-26:
Reopening because this infact does still occur:
https://www.youtube.com/watch?v=6J27M_cGxsc

InfinityGhost, 2022-03-29:
What would be a good way to go about fixing this other than what I suggested before? I'm not too fond of a disposed property and would rather use idle or out of range sort of interface if possible.

gonX, 2022-03-29:
I think this should be solved in the relevant parser if possible

vedattt, 2022-04-16:
https://github.com/X9VoiD/VoiDPlugins/issues/27

gonX, 2022-05-16:
Setting to high priority because a lot of plugin users seem to be hitting this occasionally

X9VoiD, 2022-12-31:
This is not an issue of the pipeline but rather the implementation of the output mode. Since it's already fixed I'll close this.

closed 2022-12-31 by X9VoiD