OpenTabletDriver/OpenTabletDriver #1136: Support Request: winget Support

ackhack, 2021-05-30:
# Support Request
Will there be a winget package in the future? (FYI its the new [Windows 10 PackageManager](https://github.com/microsoft/winget-cli))

adryzz, 2021-05-30:
I think there will in the near future and i think it will become the main way of installing tbh, but we'll need to make a new silent installer to make it work

sr229, 2021-06-08:
> I think there will in the near future and i think it will become the main way of installing tbh, but we'll need to make a new silent installer to make it work

Regarding that, it shouldn't be a problem, generating MSIs is done by the Winget team, you just need to [submit the manifest](https://docs.microsoft.com/en-us/windows/package-manager/package/manifest?tabs=minschema%2Cversion-example).