OpenTabletDriver/OpenTabletDriver #1138: Support Request: OpenTabletDriver not detecting GaomonS620 in MacOS

Mora502, 2021-06-01:
# Support Request
 I dont know much about programming and im kind of a noob in this sort of things, i downloaded the drivers, installed everything that was required but when i plug my Gaomon S620 to my computer it says that no tablet was found. 

## What I've already tried

## Diagnostic Information

narlock, 2021-06-01:
See here to make sure you installed OpenTabletDriver properly: https://github.com/OpenTabletDriver/OpenTabletDriver/wiki/Installation-Guide#macos-experimental

It could be the version of macOS you're running that could be causing an issue.

InfinityGhost, 2021-07-18:
Stale support request, assuming it has been resolved. This can be reopened in the case this was a mistake.

closed 2021-07-18 by InfinityGhost