OpenTabletDriver/OpenTabletDriver #1140: deleted "VEIKK" file and cannot use tablet area

rat12314, 2021-06-02:
## Description

I accidently deleted everything in the "VEIKK" file. Now opentabletdriver is no longer detecting my tablet and I cannot use my tablet area.
## System Information:

| Name             | Value |
| Rat12314 | ----- |
| Operating System  | windows 10
| Software Version | 
| Tablet           | Veikk S640

rat12314, 2021-06-02:
when i do anything, my cursor goes to the top left of my screen

jjhhgg100123, 2021-06-02:
Download the portable version and pull the Veikk files out of the configurations directory. Insert them back wherever you deleted them.

Mrcubix, 2021-06-02:
this issue shouldn't be a thing, just redownload OTD

InfinityGhost, 2021-06-03:
User error is not a problem of OpenTabletDriver.

closed 2021-06-03 by InfinityGhost

rat12314, 2021-06-03:
when i try to download OTD it doesnt download

Mrcubix, 2021-06-03:
https://www.youtube.com/watch?v=aFejLY4vKeY