OpenTabletDriver/OpenTabletDriver #1141: Huion H430P doesn't work.

rinfumi, 2021-06-03:
installed the driver and it does not respond when i move my pen.

Mrcubix, 2021-06-03:
Fill the template, no support will be provided until filled (see the template that you should had used instead of deleting it below)

```
## Description
<!-- Describe the issue below -->

## System Information:
<!-- Please fill out this information -->
| Name             | Value |
| ---------------- | ----- |
| Operating System |
| Software Version |
| Tablet           |
```

rinfumi, 2021-06-03:
> 
> ## Description
> 
> No input is done when i move my pen.
> ## System Information:
> Name 	Value
> Operating System 	Windows 10
> Software Version 	V0.5.3.1
> Tablet 	Huion H430P

re:
nevermind it fixed itself

closed 2021-06-03 by InfinityGhost