OpenTabletDriver/OpenTabletDriver #1143: Area maps incorrectly to the screen after disconnecting one monitor in a multi-monitor scenario

Kyuunex, 2021-06-04:
## Description
If I boot up my PC with 2 monitors attached, later disconnect one, the area mapped on the screen will mess up.
Same if I boot up with 1 monitor and attach another. 
For example, the tablet area will map to both screens side by side.

doing `systemctl --user restart opentabletdriver` fixes the issue for the time being.
OTD GUI also requires a restart to refresh the list of displays in the "set to display".

would be nice if this was automated somehow

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Arch Linux |
| Software Version | rolling release, everything is always latest |
| Tablet           | XP-PEN G430 |

InfinityGhost, 2021-06-04:
> rolling release, everything is always latest

Not necessarily true, still need the hash at least.

Kyuunex, 2021-06-04:
> > rolling release, everything is always latest
> 
> Not necessarily true, still need the hash at least.

specifically of what you need a version of?

Phyan, 2023-08-18:
Yeah, please do that, guys. Using more than one monitor that are constantly manually connected or using software like Spacedeck makes it impossible to comfortably use a graphics tablet, since there is no adequate mapping to the specific monitor that is used as the main workspace.