OpenTabletDriver/OpenTabletDriver #1145: Add support for XP-PEN Deco Fun series

TheVoidKitsune0, 2021-06-06:
Product Name                       | Deco Fun XS                              | Deco Fun S                                       | Deco Fun L
Product Model                      | CT430                                       | CT640                                              | CT1060
Dimensions                           | 182.4 x 134.7 x 10.5mm            | 210.5 x 159.8 x 12.9mm                   | 315 x 222.9 x 12.5mm
Active Area                           | 4.8＂x 3＂                                 | 6.3＂x 4＂                                         | 10＂x 6.27＂
Pressure Levels | 8,192
Tilt                                        | /                                                | 60°                                                    | 60°
Resolution | 5080LPI
Report Rate | ≥220RPS
Reading Height | 10mm
Compatibility | Windows 7/8/10, Mac OS X 10.10 (or later), Android 6.0 (or later), Chrome OS 88 (or later), Linux

X9VoiD, 2021-06-06:
An OTD diagnostics file are needed to add support to these tablets, will close issue until further information is provided.

> It is preferred that you create a single issue for one tablet support request and that you follow the issue template.

closed 2021-06-06 by X9VoiD

TheVoidKitsune0, 2021-06-06:
> 
> 
> An OTD diagnostics file is needed to add support to these tablets, will close issue until further information is provided.
> 
> > It is preferred that you create a single issue for one tablet support request and that you follow the issue template.
It won't let me send the .json file what should I do?

Mrcubix, 2021-06-06:
zip it and send it?

TheVoidKitsune0, 2021-06-06:
[Dia.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6604528/Dia.zip)