OpenTabletDriver/OpenTabletDriver #1146: Add support for Huion 420

Java2107, 2021-06-06:
# Tablet Information
| Name                | Value |
| --------------------- | ------- |
| Manufacturer     | Huion
| Product Name   | 420
| Width                 | 176
| Height                | 111
| Pressure Levels | 2048
| LPI                     | 4096

## Diagnostic Information
I'm unsure how to get this information as I am using Linux. Please direct me on how to do so.

## Additional information
I am requesting support for the Huion 420 (not H420...) I figured it won't be too much to add, but you never know since as far as I'm aware, the hardware for the 420 is nearly identical to the H420 (though don't quote me on this, I do think they share the same logic board) with the difference being that the hardware buttons aren't soldered on.

Mrcubix, 2021-06-06:
good thing you're on linux cause this tablet won't work on windows rn,

as for the config file, open OTD's UI, head to `help > export diagnostic`, zip that and upload it here

Java2107, 2021-06-07:
I have attached the diagnostic logs to this reply. Sorry for the delay, I was out all day yesterday.
[otd-diag.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6610705/otd-diag.txt)

X9VoiD, 2021-09-09:
Duplicate of #331

closed 2021-09-09 by X9VoiD