OpenTabletDriver/OpenTabletDriver #1151: Tablet Area doesn't reach the whole screen

mine2fs, 2021-06-07:
I got following Issue, when i set a Tablet Area in the Program the Cursor only goes to about the half of the Screen and only like 3 cm to the bottom. I know it's an Issue with my windows because i tried the Hawku Drivers and even the Wacom Drivers. When i use the full area of the Tablet it works perfectly. Does anyone have a clue what's going on? 

Btw. i have a Wacom CTL-480 and i also tried it with my XP-Pen G640

Mrcubix, 2021-06-07:
if you use the latest version & uninstall the manufacturer driver, everything should work normally, tho, every issue require a diagnostic file, like specified in the template you just removed, so i'll need you to provide one.

in the UX : help > export diagnostic > zip it > upload it

mine2fs, 2021-06-07:
Thx for the quick response.
Yeah, here ya go.
[diagnostic file.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6611218/diagnostic.file.zip)

Ferco124, 2021-06-07:
I have the same problem too
I have a CTL-4100 
[disgnostic.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/6611390/disgnostic.zip)

gonX, 2021-12-06:
Closing as a stale issue. If anyone else has the same issue, please open a new support request report. It is likely caused by old drivers conflicting.

closed 2021-12-06 by gonX