OpenTabletDriver/OpenTabletDriver #1152: Support Request: Huion H420 not detected?

yuhieki, 2021-06-07:
My Huion $30 tablet H420, I just bought from Amazon isn't working- It's not detecting it and I have it on and plugged.

My PC is a Windows 10 64 bit. I don't know what's the problem.

Mrcubix, 2021-06-07:
time to get it refunded, it's crap and isn't compatible on windows, get a S620 from gaomon's official store on aliexpress instead for the same or 2/3 the price

AbstractQbit, 2021-06-07:
Return it if you just bought it, lmao

MishterKirby, 2021-06-07:
It's not supported yet, but it should work once USB.NET is ready for primetime

lukemagdalin, 2021-06-30:
> It's not supported yet, but it should work once USB.NET is ready for primetime

What is USB.NET, I have seen it mentioned a couple times but I don't know what it is, and a google search brought nothing up.

InfinityGhost, 2021-07-01:
> What is USB.NET, I have seen it mentioned a couple times but I don't know what it is, and a google search brought nothing up.

USB.NET is a project we've been slowly working on that provides a full USB implementation in .NET from user mode permissions. The idea is that we can select devices more precisely (Device <kbd>\></kbd> Configuration <kbd>\></kbd> Interface <kbd>\></kbd> Endpoint), and with much more control.

InfinityGhost, 2021-07-18:
Duplicate of #921

closed 2021-07-18 by InfinityGhost