OpenTabletDriver/OpenTabletDriver #1162: XP-Pen Deco 1 v2 not detected in OpenTablet but registers input

jobs-git, 2021-06-10:
## Description

First of all, thank you for this software. My tablet can now register inputs! I forced remove the vendor driver `vmulti.sys` cause it is causing BSOD.

W/ OpenTabletDriver, I am not able to get inputs! But within the software it says its not detected, so I could not change the table area
## System Information:
Windows 10 v 19042.985
OpenTable v0.1.0-rc1
XP-Pen Deco 1 v2

| Name             | Value |
| ---------------- | ----- |
| Operating System |
| Software Version |
| Tablet           |

Mrcubix, 2021-06-10:
you one step away from filling the template, the most important piece of information we need to start here: the diagnostic file

[image]
zip it or .txt it

jobs-git, 2021-06-10:
Closing: outdated version

closed 2021-06-10 by jobs-git