OpenTabletDriver/OpenTabletDriver #1167: Can't download

Exaltz, 2021-06-11:
## Description
I've been trying to download it after uninstalling and when I run the installer and click install it starts then it closes and when I open it again it just does the same thing.

## System Information:
| Operating System | Windows 10
| Software Version | 21H1 I think
| Tablet           | XP-Pen Star G640S

X9VoiD, 2021-06-11:
Installer/Updater will be deprecated in the near future (it'll be integrated into OTD). If you're having problems with it just use portable instead. Closing as a `wont fix` issue.

closed 2021-06-11 by X9VoiD

Exaltz, 2021-06-11:
What is "portable?"

Exaltz, 2021-06-11:
> Installer/Updater will be deprecated in the near future (it'll be integrated into OTD). If you're having problems with it just use portable instead. Closing as a `wont fix` issue.

What is "portable?"

AbstractQbit, 2021-06-11:
https://github.com/OpenTabletDriver/OpenTabletDriver/releases