OpenTabletDriver/OpenTabletDriver #1168: Support Request:

Exaltz, 2021-06-11:
# Support Request
My cursor jumps very fast between where I am aiming and my other monitor.
https://www.youtube.com/watch?v=mLf3tLe5Vy8

## What I've already tried
Didn't find anything on the subject

## Diagnostic Information
I clicked help and export diagnostics and nothing popped up so idk

Exaltz, 2021-06-11:
This happens in osu! too but it just goes to the edge of the screen because the cursor is confined.

X9VoiD, 2021-06-11:
Uninstall any other tablet driver.

Exaltz, 2021-06-11:
They are uninstalled, I can't find anyone installed on my pc

X9VoiD, 2021-06-11:
Pretty sure there is another driver running. That behaviour only ever happens when two or more drivers are running at the same time. If you can't find it then install the official driver then uninstall it using their actual uninstaller.

Exaltz, 2021-06-11:
By official do you mean XP Pen drivers?

X9VoiD, 2021-06-11:
Yes.